mraardvark/pyupdi #11: Issue uploading program.

dingdinglhz, 2018-02-09:
I have been trying but could not get it to work.
I am using windows 10 with a standard USB-COM (CP2102 I believe) adapter, running python 3, and trying to flash a attinty1614, on a board I made myself.

Initially, I got
File "C:\Users\linhz\Downloads\pyupdi-master\updi\physical.py", line 103, in receive
response.append(ord(character))
TypeError: ord() expected a character, but string of length 0 found

And the command I used is
python pyupdi.py -d tiny1614 -c COM4 -f "XXXX path to the .hex"

Then I observed that when I turn on the system and run it the first time, it seems to go a bit further. I also suspected a 100k pull-up resistor on my rst line might screw up the interface, so I removed it. I also turned on verbose to get the most info.

The first attempt after power on gives me:
`INFO:phy Opening COM4 at 9600 baud
INFO:phy send : [0x0]
INFO:link STCS to 0x03
INFO:phy send : [0x55, 0xc3, 0x8]
INFO:link STCS to 0x02
INFO:phy send : [0x55, 0xc2, 0x80]
INFO:link LDCS from 0x00
INFO:phy send : [0x55, 0x80]
INFO:phy receive : [0x20]
INFO:link UPDI init OK
UPDI programmer created successfully
INFO:nvm Entering NVM programming mode
INFO:link LDCS from 0x0B
INFO:phy send : [0x55, 0x8b]
INFO:phy receive : [0x82]
INFO:app Entering NVM programming mode
INFO:link Writing key
INFO:phy send : [0x55, 0xe0]
INFO:phy send : [0x20, 0x67, 0x6f, 0x72, 0x50, 0x4d, 0x56, 0x4e]
Device is locked. Performing unlock with chip erase.
INFO:link Writing key
INFO:phy send : [0x55, 0xe0]
INFO:phy send : [0x65, 0x73, 0x61, 0x72, 0x45, 0x4d, 0x56, 0x4e]
Traceback (most recent call last):
  File "pyupdi.py", line 88, in _main
    nvm.enter_progmode()
  File "C:\Users\linhz\Downloads\pyupdi-master\updi\nvm.py", line 30, in enter_progmode
    if self.application.enter_progmode():
  File "C:\Users\linhz\Downloads\pyupdi-master\updi\application.py", line 96, in enter_progmode
    self.datalink.key(constants.UPDI_KEY_64, constants.UPDI_KEY_NVM)
  File "C:\Users\linhz\Downloads\pyupdi-master\updi\link.py", line 204, in key
    self.updi_phy.send(list(reversed(list(key))))
  File "C:\Users\linhz\Downloads\pyupdi-master\updi\physical.py", line 80, in send
    self.ser.write([character])
  File "C:\ProgramData\Anaconda3\lib\site-packages\serial\serialwin32.py", line 308, in write
    data = to_bytes(data)
  File "C:\ProgramData\Anaconda3\lib\site-packages\serial\serialutil.py", line 66, in to_bytes
    return bytes(bytearray(seq))
TypeError: an integer is required

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "pyupdi.py", line 150, in <module>
    _main()
  File "pyupdi.py", line 91, in _main
    nvm.unlock_device()
  File "C:\Users\linhz\Downloads\pyupdi-master\updi\nvm.py", line 50, in unlock_device
    self.application.unlock()
  File "C:\Users\linhz\Downloads\pyupdi-master\updi\application.py", line 67, in unlock
    self.datalink.key(constants.UPDI_KEY_64, constants.UPDI_KEY_CHIPERASE)
  File "C:\Users\linhz\Downloads\pyupdi-master\updi\link.py", line 204, in key
    self.updi_phy.send(list(reversed(list(key))))
  File "C:\Users\linhz\Downloads\pyupdi-master\updi\physical.py", line 80, in send
    self.ser.write([character])
  File "C:\ProgramData\Anaconda3\lib\site-packages\serial\serialwin32.py", line 308, in write
    data = to_bytes(data)
  File "C:\ProgramData\Anaconda3\lib\site-packages\serial\serialutil.py", line 66, in to_bytes
    return bytes(bytearray(seq))
TypeError: an integer is required
INFO:phy Closing COM4`

And the subsequent times:
`INFO:phy Opening COM4 at 9600 baud
INFO:phy send : [0x0]
INFO:link STCS to 0x03
INFO:phy send : [0x55, 0xc3, 0x8]
INFO:link STCS to 0x02
INFO:phy send : [0x55, 0xc2, 0x80]
INFO:link LDCS from 0x00
INFO:phy send : [0x55, 0x80]
INFO:phy receive : [0x20]
INFO:link UPDI init OK
UPDI programmer created successfully
INFO:nvm Entering NVM programming mode
INFO:link LDCS from 0x0B
INFO:phy send : [0x55, 0x8b]
INFO:phy receive : [0x82]
INFO:app Entering NVM programming mode
INFO:link Writing key
INFO:phy send : [0x55, 0xe0]
INFO:phy send : [0x20, 0x67, 0x6f, 0x72, 0x50, 0x4d, 0x56, 0x4e]
Device is locked. Performing unlock with chip erase.
INFO:link Writing key
INFO:phy send : [0x55, 0xe0]
INFO:phy send : [0x65, 0x73, 0x61, 0x72, 0x45, 0x4d, 0x56, 0x4e]
Traceback (most recent call last):
  File "pyupdi.py", line 88, in _main
    nvm.enter_progmode()
  File "C:\Users\linhz\Downloads\pyupdi-master\updi\nvm.py", line 30, in enter_progmode
    if self.application.enter_progmode():
  File "C:\Users\linhz\Downloads\pyupdi-master\updi\application.py", line 96, in enter_progmode
    self.datalink.key(constants.UPDI_KEY_64, constants.UPDI_KEY_NVM)
  File "C:\Users\linhz\Downloads\pyupdi-master\updi\link.py", line 204, in key
    self.updi_phy.send(list(reversed(list(key))))
  File "C:\Users\linhz\Downloads\pyupdi-master\updi\physical.py", line 80, in send
    self.ser.write([character])
  File "C:\ProgramData\Anaconda3\lib\site-packages\serial\serialwin32.py", line 308, in write
    data = to_bytes(data)
  File "C:\ProgramData\Anaconda3\lib\site-packages\serial\serialutil.py", line 66, in to_bytes
    return bytes(bytearray(seq))
TypeError: an integer is required

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "pyupdi.py", line 150, in <module>
    _main()
  File "pyupdi.py", line 91, in _main
    nvm.unlock_device()
  File "C:\Users\linhz\Downloads\pyupdi-master\updi\nvm.py", line 50, in unlock_device
    self.application.unlock()
  File "C:\Users\linhz\Downloads\pyupdi-master\updi\application.py", line 67, in unlock
    self.datalink.key(constants.UPDI_KEY_64, constants.UPDI_KEY_CHIPERASE)
  File "C:\Users\linhz\Downloads\pyupdi-master\updi\link.py", line 204, in key
    self.updi_phy.send(list(reversed(list(key))))
  File "C:\Users\linhz\Downloads\pyupdi-master\updi\physical.py", line 80, in send
    self.ser.write([character])
  File "C:\ProgramData\Anaconda3\lib\site-packages\serial\serialwin32.py", line 308, in write
    data = to_bytes(data)
  File "C:\ProgramData\Anaconda3\lib\site-packages\serial\serialutil.py", line 66, in to_bytes
    return bytes(bytearray(seq))
TypeError: an integer is required
INFO:phy Closing COM4`

Any hints? I kind of need to fix this issues soon. Any help greatly appreciated. Thanks in advance.

mraardvark, 2018-02-10:
On power-up UPDI is disabled, and pyupdi sends a break to enable it.  On subsequent runs without power toggle, this is interpreted as a start bit, and then a double-break is used to recover.  This explains the differences after power up.

I have not tested on python3 at all - perhaps try py27, or try one of the py3 forks of pyupdi?

dingdinglhz, 2018-02-10:
I tested with pyhon27 and it works. I do remember vaguely that pyserial works slightly differently between py27 and py3. Thanks. 
If I have time I might look into it.

HalexRomagno, 2018-05-29:
Hi @dingdinglhz , have you had the time to look into what to change in the code to make it work with Py3?

linhz0hz, 2018-09-20:
Unfortunately no

mraardvark, 2018-09-20:
Py3 was working last time I tried: https://github.com/mraardvark/pyupdi/pull/14

closed 2018-10-22 by mraardvark