mraardvark/pyupdi #21: Support for megaAVR 0-series

fschneidereit, 2018-10-30:
Support for the ATmega3208/3209/4808/4809 would be great.

mraardvark, 2018-10-30:
Technically UPDI is the same on the megas, but the flash starts at 0x4000.

fschneidereit, 2018-10-30:
I don't work with these devices at the moment, but would contribute the necessary code if flash address is the only difference. What about the ATtiny3216/3217? Telling from the data sheet, its the same as all the other tinyAVR 1-series devices, except for flash size.

mraardvark, 2018-10-30:
I have a mega4809 here, but need to do a spot of soldering to use pyupdi...  will note-to-self to try to get it done...

fschneidereit, 2018-10-30:
OK. The page size is also different on the megas (128). I pushed a version with megaAVR support to my fork of pyupdi, you can use it for experimenting if you like.

mraardvark, 2018-10-30:
Thanks.  I did a cut-and-paste over and pushed.  Tested with led blinker on m4809.

closed 2018-10-30 by mraardvark