mraardvark/pyupdi #23: ATTiny402 Exception: Timeout waiting for flash ready after page write

odwdinc, 2018-12-31:
trying to program ATtiny402 
seem to be able to run 
python3 pyupdi.py -d tiny402 -c /dev/ttyUSB0  -e
to Erase the chip
but when running

python3 pyupdi.py -d tiny402 -c /dev/ttyUSB0 -v -f SbusSwitchATtiny.hex       

INFO:phy Opening /dev/ttyUSB0 at 115200 baud
INFO:phy send : [0x0]
INFO:link STCS to 0x03
INFO:phy send : [0x55, 0xc3, 0x8]
INFO:link STCS to 0x02
INFO:phy send : [0x55, 0xc2, 0x80]
INFO:link LDCS from 0x00
INFO:phy send : [0x55, 0x80]
INFO:phy receive : []
INFO:link UPDI not OK - reinitialisation required
INFO:phy Sending double break
INFO:phy Opening /dev/ttyUSB0 at 115200 baud
INFO:link STCS to 0x03
INFO:phy send : [0x55, 0xc3, 0x8]
INFO:link STCS to 0x02
INFO:phy send : [0x55, 0xc2, 0x80]
INFO:link LDCS from 0x00
INFO:phy send : [0x55, 0x80]
INFO:phy receive : [0x20]
INFO:link UPDI init OK
INFO:nvm Entering NVM programming mode
INFO:link LDCS from 0x0B
INFO:phy send : [0x55, 0x8b]
INFO:phy receive : [0x82]
INFO:app Entering NVM programming mode
INFO:link Writing key
INFO:phy send : [0x55, 0xe0]
INFO:phy send : [0x20, 0x67, 0x6f, 0x72, 0x50, 0x4d, 0x56, 0x4e]
INFO:link LDCS from 0x07
INFO:phy send : [0x55, 0x87]
INFO:phy receive : [0x10]
INFO:app Key status = 0x10
INFO:app Apply reset
INFO:link STCS to 0x08
INFO:phy send : [0x55, 0xc8, 0x59]
INFO:app Release reset
INFO:link STCS to 0x08
INFO:phy send : [0x55, 0xc8, 0x0]
INFO:link LDCS from 0x0B
INFO:phy send : [0x55, 0x8b]
INFO:phy receive : [0x21]
INFO:link LDCS from 0x0B
INFO:phy send : [0x55, 0x8b]
INFO:phy receive : [0x8]
INFO:link LDCS from 0x0B
INFO:phy send : [0x55, 0x8b]
INFO:phy receive : [0x8]
INFO:app Now in NVM programming mode
INFO:nvm Reading device info
INFO:phy send : [0x55, 0xe5]
INFO:app SIB read out as: b'tinyAVR P:0D:0-3'
INFO:app Family ID = b'tinyAVR'
INFO:app NVM revision = 48
INFO:app OCD revision = 48
INFO:app PDI OSC = 51MHz
INFO:link LDCS from 0x00
INFO:phy send : [0x55, 0x80]
INFO:phy receive : [0x20]
INFO:app PDI revision = 0x2
INFO:link LDCS from 0x0B
INFO:phy send : [0x55, 0x8b]
INFO:phy receive : [0x8]
INFO:app Reading 3 bytes from 0x1100
INFO:link ST to ptr
INFO:phy send : [0x55, 0x69, 0x0, 0x11]
INFO:phy receive : [0x40]
INFO:link Repeat 3
INFO:phy send : [0x55, 0xa1, 0x2, 0x0]
INFO:link LD8 from ptr++
INFO:phy send : [0x55, 0x24]
INFO:phy receive : [0x1e, 0x92, 0x25]
INFO:app Reading 1 bytes from 0x0F01
INFO:link ST to ptr
INFO:phy send : [0x55, 0x69, 0x1, 0xf]
INFO:phy receive : [0x40]
INFO:link LD8 from ptr++
INFO:phy send : [0x55, 0x24]
INFO:phy receive : [0x1]
INFO:app Device ID = 1E9225 rev B
INFO:nvm Loading from hexfile 'SbusSwitchATtiny.hex'
INFO:nvm Loaded 3317 bytes from ihex starting at address 0x2000
INFO:nvm Adjusting flash offset to address 0x8000
INFO:app Chip erase using NVM CTRL
INFO:app Wait flash ready
INFO:link LD from 0x1002
INFO:phy send : [0x55, 0x4, 0x2, 0x10]
INFO:phy receive : [0x0]
INFO:app NVMCMD 5 executing
INFO:link ST to 0x1000
INFO:phy send : [0x55, 0x44, 0x0, 0x10]
INFO:phy receive : [0x40]
INFO:phy send : [0x5]
INFO:phy receive : [0x40]
INFO:app Wait flash ready
INFO:link LD from 0x1002
INFO:phy send : [0x55, 0x4, 0x2, 0x10]
INFO:phy receive : [0x0]
INFO:nvm Padding to blocksize 64 with 0xFF
INFO:nvm Paging into 64 byte blocks
INFO:nvm Writing page at 0xA000
INFO:app Wait flash ready
INFO:link LD from 0x1002
INFO:phy send : [0x55, 0x4, 0x2, 0x10]
INFO:phy receive : [0x0]
INFO:app Clear page buffer
INFO:app NVMCMD 4 executing
INFO:link ST to 0x1000
INFO:phy send : [0x55, 0x44, 0x0, 0x10]
INFO:phy receive : [0x40]
INFO:phy send : [0x4]
INFO:phy receive : [0x40]
INFO:app Wait flash ready
INFO:link LD from 0x1002
INFO:phy send : [0x55, 0x4, 0x2, 0x10]
INFO:phy receive : [0x0]
INFO:link ST to ptr
INFO:phy send : [0x55, 0x69, 0x0, 0xa0]
INFO:phy receive : [0x40]
INFO:link Repeat 32
INFO:phy send : [0x55, 0xa1, 0x1f, 0x0]
INFO:link ST16 to *ptr++
INFO:phy send : [0x55, 0x65, 0x19, 0xc0]
INFO:phy receive : [0x40]
INFO:phy send : [0x33, 0xc0]
INFO:phy receive : [0x40]
INFO:phy send : [0x32, 0xc0]
INFO:phy receive : [0x40]
INFO:phy send : [0x31, 0xc0]
INFO:phy receive : [0x40]
INFO:phy send : [0x30, 0xc0]
INFO:phy receive : [0x40]
INFO:phy send : [0x2f, 0xc0]
INFO:phy receive : [0x40]
INFO:phy send : [0x2e, 0xc0]
INFO:phy receive : [0x40]
INFO:phy send : [0x2d, 0xc0]
INFO:phy receive : [0x40]
INFO:phy send : [0x2c, 0xc0]
INFO:phy receive : [0x40]
INFO:phy send : [0x2b, 0xc0]
INFO:phy receive : [0x40]
INFO:phy send : [0x2a, 0xc0]
INFO:phy receive : [0x40]
INFO:phy send : [0x29, 0xc0]
INFO:phy receive : [0x40]
INFO:phy send : [0x28, 0xc0]
INFO:phy receive : [0x40]
INFO:phy send : [0x27, 0xc0]
INFO:phy receive : [0x40]
INFO:phy send : [0x26, 0xc0]
INFO:phy receive : [0x40]
INFO:phy send : [0x25, 0xc0]
INFO:phy receive : [0x40]
INFO:phy send : [0x24, 0xc0]
INFO:phy receive : [0x40]
INFO:phy send : [0x23, 0xc0]
INFO:phy receive : [0x40]
INFO:phy send : [0x22, 0xc0]
INFO:phy receive : [0x40]
INFO:phy send : [0x21, 0xc0]
INFO:phy receive : [0x40]
INFO:phy send : [0x20, 0xc0]
INFO:phy receive : [0x40]
INFO:phy send : [0x1f, 0xc0]
INFO:phy receive : [0x40]
INFO:phy send : [0x4f, 0xc5]
INFO:phy receive : [0x40]
INFO:phy send : [0x77, 0xc5]
INFO:phy receive : [0x40]
INFO:phy send : [0x1c, 0xc0]
INFO:phy receive : [0x40]
INFO:phy send : [0x1b, 0xc0]
INFO:phy receive : [0x40]
INFO:phy send : [0x11, 0x24]
INFO:phy receive : [0x40]
INFO:phy send : [0x1f, 0xbe]
INFO:phy receive : [0x40]
INFO:phy send : [0xcf, 0xef]
INFO:phy receive : [0x40]
INFO:phy send : [0xcd, 0xbf]
INFO:phy receive : [0x40]
INFO:phy send : [0xdf, 0xe3]
INFO:phy receive : [0x40]
INFO:phy send : [0xde, 0xbf]
INFO:phy receive : [0x40]
INFO:app Committing page
INFO:app NVMCMD 1 executing
INFO:link ST to 0x1000
INFO:phy send : [0x55, 0x44, 0x0, 0x10]
INFO:phy receive : [0x40]
INFO:phy send : [0x1]
INFO:phy receive : [0x40]
INFO:app Wait flash ready
INFO:link LD from 0x1002
INFO:phy send : [0x55, 0x4, 0x2, 0x10]
INFO:phy receive : [0x4]
INFO:app NVM error
Traceback (most recent call last):
  File "pyupdi.py", line 157, in <module>
    _main()
  File "pyupdi.py", line 98, in _main
    if not _process(nvm, args):
  File "pyupdi.py", line 122, in _process
    return _flash_file(nvm, args.flash)
  File "pyupdi.py", line 132, in _flash_file
    nvm.write_flash(start_address, data)
  File "./pyupdi/updi/nvm.py", line 106, in write_flash
    self.application.write_nvm(address, page)
  File "./pyupdi/updi/application.py", line 267, in write_nvm
    raise Exception("Timeout waiting for flash ready after page write ")
INFO:phy Closing /dev/ttyUSB0

mraardvark, 2019-01-05:
The device [family] selection mechanism has issues, so the flash page size is 64 instead of 32.  I will try to fix this soon, but in the mean time you can change it locally to 32b in device.py

mraardvark, 2019-01-05:
ATtiny402 has 4K FLASH.
Your hexfile has 3317 bytes (ok) but it has an offset of 0x2000:
    "Loaded 3317 bytes from ihex starting at address 0x2000"
This puts it outside the range for this part.
Where did you get this hexfile from?  Or is there some custom linker or post-build which adds this offset?

mraardvark, 2019-01-11:
Strange. Any chance you could share the project, or a minimal setup which has the same mystery-offset?

odwdinc, 2019-01-24:
i think it was the project. others are working, sorry i did not debug more on my end first :(

closed 2019-01-24 by odwdinc