mraardvark/pyupdi #24: SIB read out appears abnormal

cmcraft, 2019-03-10:
Maybe I'm misusing the application, but I'm running into the following error when I attempt to program an ATtiny1614 via a Teensy 3.2 as a serial adapter.

Error below:
```
INFO:phy Opening COM8 at 115200 baud
INFO:phy send : [0x0]
INFO:link STCS to 0x03
INFO:phy send : [0x55, 0xc3, 0x8]
INFO:link STCS to 0x02
INFO:phy send : [0x55, 0xc2, 0x80]
INFO:link LDCS from 0x00
INFO:phy send : [0x55, 0x80]
INFO:phy receive : [0x30]
INFO:link UPDI init OK
INFO:nvm Entering NVM programming mode
INFO:link LDCS from 0x0B
INFO:phy send : [0x55, 0x8b]
INFO:phy receive : [0x35]
INFO:app Entering NVM programming mode
INFO:link Writing key
INFO:phy send : [0x55, 0xe0]
INFO:phy send : [0x20, 0x67, 0x6f, 0x72, 0x50, 0x4d, 0x56, 0x4e]
INFO:link LDCS from 0x07
INFO:phy send : [0x55, 0x87]
INFO:phy receive : [0x30]
INFO:app Key status = 0x30
INFO:app Apply reset
INFO:link STCS to 0x08
INFO:phy send : [0x55, 0xc8, 0x59]
INFO:app Release reset
INFO:link STCS to 0x08
INFO:phy send : [0x55, 0xc8, 0x0]
INFO:link LDCS from 0x0B
INFO:phy send : [0x55, 0x8b]
INFO:phy receive : [0xd]
INFO:link LDCS from 0x0B
INFO:phy send : [0x55, 0x8b]
INFO:phy receive : [0x33]
INFO:link LDCS from 0x0B
INFO:phy send : [0x55, 0x8b]
INFO:phy receive : [0x31]
INFO:link LDCS from 0x0B
INFO:phy send : [0x55, 0x8b]
INFO:phy receive : [0xa]
INFO:link LDCS from 0x0B
INFO:phy send : [0x55, 0x8b]
INFO:phy receive : [0xd]
INFO:app Now in NVM programming mode
INFO:nvm Reading device info
INFO:phy send : [0x55, 0xe5]
INFO:app SIB read out as: b'6\r\n'
INFO:app Family ID = b'6\r\n'
Traceback (most recent call last):
  File "C:\Users\NeruA\Desktop\ATtiny1614 programming\pyupdi-master\pyupdi.py", line 157, in <module>
    _main()
  File "C:\Users\NeruA\Desktop\ATtiny1614 programming\pyupdi-master\pyupdi.py", line 96, in _main
    nvm.get_device_info()
  File "C:\Users\NeruA\Desktop\ATtiny1614 programming\pyupdi-master\updi\nvm.py", line 27, in get_device_info
    return self.application.device_info()
  File "C:\Users\NeruA\Desktop\ATtiny1614 programming\pyupdi-master\updi\application.py", line 29, in device_info
    self.logger.info("NVM revision = {}".format(sib[10]))
IndexError: index out of range
INFO:phy Closing COM8
```

Teensy configuration:

```
#define Uart Serial2

void setup()
{
  Uart.begin(115200);
  Serial.begin(115200);
}

void loop() {
  byte usb_data;
  byte uart_data;
  if (Serial.available() > 0) {
    usb_data = Serial.read();
    Uart.println(usb_data);
  }
    

  if (Uart.available() > 0) {
    uart_data = Uart.read();
    Serial.println(uart_data);
  }
}
```

Wiring: 

Teensy | intermediates | ATtiny1614
-- | -- | --
GND | -- | GND on 14
Rx2 (Pin 9) | -- | UPDI on 10
Tx2 (Pin 10) | 4k7 resistor | UPDI on 10
3.3v | -- | VDD on 1

cmcraft, 2019-03-10:
When I attempted to use write/print the UPDI interface failed to initialize so I assumed something in pyUPDI was looking for a \n as a break.

> Here's the error when I switch to write.
> 
> INFO:phy Opening COM8 at 115200 baud
> INFO:phy send : [0x0]
> INFO:link STCS to 0x03
> INFO:phy send : [0x55, 0xc3, 0x8]
> INFO:link STCS to 0x02
> INFO:phy send : [0x55, 0xc2, 0x80]
> INFO:link LDCS from 0x00
> INFO:phy send : [0x55, 0x80]
> INFO:phy receive : []
> INFO:link UPDI not OK - reinitialisation required
> INFO:phy Sending double break
> INFO:phy Opening COM8 at 115200 baud
> INFO:link STCS to 0x03
> INFO:phy send : [0x55, 0xc3, 0x8]
> INFO:link STCS to 0x02
> INFO:phy send : [0x55, 0xc2, 0x80]
> INFO:link LDCS from 0x00
> INFO:phy send : [0x55, 0x80]
> INFO:phy receive : []
> INFO:link UPDI not OK - reinitialisation required
> Traceback (most recent call last):
> File "C:\Users\NeruA\Desktop\ATtiny1614 programming\pyupdi-master\pyupdi.py", line 157, in <module>
> _main()
> File "C:\Users\NeruA\Desktop\ATtiny1614 programming\pyupdi-master\pyupdi.py", line 88, in _main
> device=Device(args.device))
> File "C:\Users\NeruA\Desktop\ATtiny1614 programming\pyupdi-master\updi\nvm.py", line 17, in __init__
> self.application = UpdiApplication(comport, baud, device)
> File "C:\Users\NeruA\Desktop\ATtiny1614 programming\pyupdi-master\updi\application.py", line 17, in __init__
> self.datalink = UpdiDatalink(comport, baud)
> File "C:\Users\NeruA\Desktop\ATtiny1614 programming\pyupdi-master\updi\link.py", line 30, in __init__
> raise Exception("UPDI initialisation failed")
> Exception: UPDI initialisation failed
> INFO:phy Closing COM8

cmcraft, 2019-03-13:
After using a proper USB to TTL Serial adapter things went fine!

closed 2019-03-13 by cmcraft

mraardvark, 2019-03-13:
Ah, good to hear it.