mraardvark/pyupdi #27: Running on Raspberry PI

becky001, 2019-07-17:
I am running on Raspberry PI using GPIO 14 & 15, i.e. pins 8,10, using 4.7k resistor between TX and RX.  I get the following output:  

 $ python pyupdi.py -d tiny814 -c /dev/ttyAMA0 -e -f firmware.hex -v
INFO:phy Opening /dev/ttyAMA0 at 115200 baud
INFO:phy send : [0x0]
INFO:phy incorrect echo : [0x0]
Traceback (most recent call last):
  File "pyupdi.py", line 160, in <module>
    _main()
  File "pyupdi.py", line 90, in _main
    device=Device(args.device))
  File "/home/pi/Documents/PY/updi/nvm.py", line 17, in __init__
    self.application = UpdiApplication(comport, baud, device)
  File "/home/pi/Documents/PY/updi/application.py", line 17, in __init__
    self.datalink = UpdiDatalink(comport, baud)
  File "/home/pi/Documents/PY/updi/link.py", line 20, in __init__
    self.updi_phy = UpdiPhysical(comport, baud)
  File "/home/pi/Documents/PY/updi/physical.py", line 30, in __init__
    self.send([constants.UPDI_BREAK])
  File "/home/pi/Documents/PY/updi/physical.py", line 85, in send
    raise Exception("Incorrect echo data")
Exception: Incorrect echo data

I checked the Tx and Rx  line with a scope and confirmed a 88usec logic "0" pulse, which I think is the 0x0 being sent at 115200 baud.  But the output says:  INFO:phy incorrect echo : [0x0].  Why is that incorrect echo ?  A 0x0 is sent, the echo is 0x0.

PitterL, 2019-09-15:
I think it's difficult to debug embedded system for python, special for I/O timing, if you has c basic, you could think about cupdi lib alternative.

mraardvark, 2019-09-26:
are you using python3??

becky001, 2019-09-26:
No.  I am running Python 2.

xoriath, 2019-09-27:
Try with python3, ref #32 and #33

becky001, 2020-01-16:
Remove the echo checking lines in physical.py and it works.  Essentially I backed out the changes to physical.py for "Performance improvement" March 2019.  
Works great now on Raspi.

closed 2020-01-16 by becky001