mraardvark/pyupdi #30: Verify Flash after Programming

mikrocoder, 2019-08-21:
Hello,

may I ask how the programmer works internally? Normally there is a read process after writing to check. "Verify Flash after Programming". 
The Tx data are however passed on as far as I see 1:1 to the Rx pin by means of a resistor. Is it possible to install a real read test? Or is that withheld from the hardware programmers?

PitterL, 2019-09-15:
The protocol is UPDI, which is a serial derived protocol, we use serial port combine TX/RX to 1 line, to simulate the real protocol. I suggest you check the chip datasheet for communication details.

avian2, 2020-03-12:
In case anyone else is wondering whether `pyupdi` verifies flash contents after writing and found this issue: the answer is yes, it does so by default when using the `-f`/`--flash` flag. UPDI is a two-way protocol. Relevant code for verifying flash contents after programming is done [starts here](https://github.com/mraardvark/pyupdi/blob/907106c5e979f41c6068a5a67b8ca8b330f9af45/pyupdi.py#L146).