mraardvark/pyupdi #32: Can't get it to work?

silardg, 2019-09-02:
Hi!

I'm getting error:

`INFO:phy Opening /dev/tty.SLAB_USBtoUART at 115200 baud
INFO:phy send : [0x0]
INFO:phy incorrect echo : [0x0]
Traceback (most recent call last):
  File "pyupdi.py", line 160, in <module>
    _main()
  File "pyupdi.py", line 90, in _main
    device=Device(args.device))
  File "/Users/silardgal/Documents/Dev/SENSA/pecka_pcb/Software/pyupdi/updi/nvm.py", line 17, in __init__
    self.application = UpdiApplication(comport, baud, device)
  File "/Users/silardgal/Documents/Dev/SENSA/pecka_pcb/Software/pyupdi/updi/application.py", line 17, in __init__
    self.datalink = UpdiDatalink(comport, baud)
  File "/Users/silardgal/Documents/Dev/SENSA/pecka_pcb/Software/pyupdi/updi/link.py", line 20, in __init__
    self.updi_phy = UpdiPhysical(comport, baud)
  File "/Users/silardgal/Documents/Dev/SENSA/pecka_pcb/Software/pyupdi/updi/physical.py", line 30, in __init__
    self.send([constants.UPDI_BREAK])
  File "/Users/silardgal/Documents/Dev/SENSA/pecka_pcb/Software/pyupdi/updi/physical.py", line 85, in send
    raise Exception("Incorrect echo data")
Exception: Incorrect echo data
INFO:phy Closing /dev/tty.SLAB_USBtoUART`

Any idea? I have everything soldered like in the readme.

silardg, 2019-09-02:
Progress!!!

I have change it to python3 and used 3K resistor (as for someone it worked on another issue)

`python3 pyupdi.py -c /dev/tty.SLAB_USBtoUART -d mega1609 -r -v -b 115200
INFO:phy Opening /dev/tty.SLAB_USBtoUART at 115200 baud
INFO:phy send : [0x0]
INFO:link STCS to 0x03
INFO:phy send : [0x55, 0xc3, 0x8]
INFO:link STCS to 0x02
INFO:phy send : [0x55, 0xc2, 0x80]
INFO:link LDCS from 0x00
INFO:phy send : [0x55, 0x80]
INFO:phy receive : []
INFO:link UPDI not OK - reinitialisation required
INFO:phy Sending double break
INFO:phy Opening /dev/tty.SLAB_USBtoUART at 115200 baud
INFO:link STCS to 0x03
INFO:phy send : [0x55, 0xc3, 0x8]
INFO:link STCS to 0x02
INFO:phy send : [0x55, 0xc2, 0x80]
INFO:link LDCS from 0x00
INFO:phy send : [0x55, 0x80]
INFO:phy receive : []
INFO:link UPDI not OK - reinitialisation required
Traceback (most recent call last):
  File "pyupdi.py", line 160, in <module>
    _main()
  File "pyupdi.py", line 90, in _main
    device=Device(args.device))
  File "/Users/silardgal/Documents/Dev/SENSA/pecka_pcb/Software/pyupdi/updi/nvm.py", line 17, in __init__
    self.application = UpdiApplication(comport, baud, device)
  File "/Users/silardgal/Documents/Dev/SENSA/pecka_pcb/Software/pyupdi/updi/application.py", line 17, in __init__
    self.datalink = UpdiDatalink(comport, baud)
  File "/Users/silardgal/Documents/Dev/SENSA/pecka_pcb/Software/pyupdi/updi/link.py", line 31, in __init__
    raise Exception("UPDI initialisation failed")
Exception: UPDI initialisation failed
INFO:phy Closing /dev/tty.SLAB_USBtoUART`

silardg, 2019-09-02:
To add things I have tried:
- 3K and 4K7 resistor
- removed pull up on reset 
- changed baud

And nothing has worked, I can't get a receive back.

MAMARR, 2019-09-03:
Hi, we are using python 2.7.9 and trying to flash Atmesga4809, seeing similar issue, please let us know

C:\Python27>python pyupdi.py -d mega4809 -c COM6 -b 115200 -e -v -f binn.hex
INFO:phy Opening COM6 at 115200 baud
INFO:phy send : [0x0]
INFO:phy incorrect echo : [0x0]
Traceback (most recent call last):
  File "pyupdi.py", line 165, in <module>
    _main()
  File "pyupdi.py", line 95, in _main
    device=Device(args.device))
  File "C:\Python27\nvm.py", line 18, in __init__
    self.application = UpdiApplication(comport, baud, device)
  File "C:\Python27\application.py", line 20, in __init__
    self.datalink = UpdiDatalink(comport, baud)
  File "C:\Python27\link.py", line 22, in __init__
    self.updi_phy = UpdiPhysical(comport, baud)
  File "C:\Python27\physical.py", line 30, in __init__
    self.send([constants.UPDI_BREAK])
  File "C:\Python27\physical.py", line 85, in send
    raise Exception("Incorrect echo data")
Exception: Incorrect echo data
INFO:phy Closing COM6

PitterL, 2019-09-15:
Looks like the communiction couldn't be build between your UART chip and MCU. You should check whether the UART chip match:
parity=serial.PARITY_EVEN ,stopbits=serial.STOPBITS_TWO

paulbonnett, 2019-09-25:
I have the same problem, tried it with CP2102 and FT232R bridge on both Linux and Windows 10 and different baud rates. Still the same problem.  Using a serial terminal characters are echoed OK.
Any ideas?

mraardvark, 2019-09-25:
If you are getting an proper echo (round trip) and not a local echo with a terminal, then the problem could still be in a few places... :/
- python3?
- correct serial access rights (linux)?
- recent pyserial?
- sure you have not disabled UPDI (high-voltage required)?
I will have to dig out my setup and give it a spin again one day...

paulbonnett, 2019-09-26:
Thanks for prompt reply.
Checked for local echo by removing resistor, no local echo
Python version 2.7.15+
Yes to serial access rights.
pyserial 3.4
Tested target with jtag2updi, which worked, so UPDI not disabled

mraardvark, 2019-09-26:
Probably not the issue here, but you really should be heading towards py3 :|

paulbonnett, 2019-09-26:
Actually it was the issue, tried it with py3 and it worked.  Thanks a lot for your help and a great utility

mraardvark, 2019-09-26:
Good to hear it :)

closed 2019-09-26 by mraardvark