mraardvark/pyupdi #36: Interface board for UPDI microprocessors

HannesBulk, 2019-11-01:
I found your program, and based on this idee i devellopped a small interface board to be used as a Serial interface / Programmer for all processors with a UPDI interface.
It is basic, does not support High Voltage reset for the ATTiny devices in case the UPDI pin is fused to be used for i/o or reset, but it can be used in to modes: USB Serial device, for sending/receiving data, and Programming mode: Used to program a UPDI microprocessor.
The principle is very simple: After power up, the device acts as a normal serial interface, it even handles the RTS line (not DTR) to generate a reset puls for Arduino style boards. 
To put the board in Programming mode, a special sequence must be followed:
1.	Open the serial port, with RTS Off (+5v). 
2.	Start a BREAK
3.	Set RTS ON (0 volt)
4.	Clear break
The serial port pins are now connected to the UPDI interface, and programming can be done.
To end programming simply set RTS OFF (+5Volt), this will end programming mode, and will reactivate the normal serial interface function.
I included a ZIP file with the schematics and a short explanation. If you are interested I can also send you a working prototype of the PCB. (Still working om improvements, but this prototype is tested and it works. In order to make this board fully supported by your program some extra code has to be put in the program to handle the programming mode sequence and end-of-program sequence. This will not effect the working of your program.
More details in the attached ZIP file

[UPDI Project.zip](https://github.com/mraardvark/pyupdi/files/3797786/UPDI.Project.zip)

mraardvark, 2019-11-08:
Very clever @HannesBulk :)
I do however think that you should put any changes required by your hardware into a fork of pyupdi rather than merge them into master here.  I would like to keep this as simple as possible (and tested :)

HannesBulk, 2019-11-12:
Are you interested in the prototype ? Send me a private mail for details, and you'll get a working example.

mraardvark, 2019-12-06:
Sorry @HannesBulk, I guess you can tell I am limited in time right now, but thanks for the offer!