mraardvark/pyupdi #40: Ways to use through Arduino IDE

SpenceKonde, 2019-12-17:
Hi

I maintain megaTinyCore ( https://github.com/SpenceKonde/megaTinyCore ) which lets these parts be used through the Arduino IDE. Right now we are telling people to use a jtag2updi firmware on a nano/promini/etc to program these parts. This looks like a great solution in terms of not needing additional hardware, just a serial adapter. I would like to make it possible to use this programming method from the Arduino IDE - I see two approaches.

Am I correct in concluding that we can't bully AVRDude into doing this?

That would mean that we would, instead need to create a new "tool" for this. Is there a way to make a self-contained version of pyupdi that board manager could pull in with tools_dependencies (ie, will work without having to install a python runtime or anything else)? my understanding is that if that were possible, I could have my core pull that in upon installation and have a programmer option that would use pyupdi to upload with just a serial port. 

Thanks

mraardvark, 2019-12-17:
I think avrdude is no longer the way to go, and adding a new tool in the arduino ide is a better approach.
But without native python its necessary to go for a "py2exe"-like approach which then needs more OS support.  I have not done this in a while so I will need to explore some options first.
But yes, this would be cool :)

mikkel75, 2020-09-17:
Anyone working on making pyupdi supported by Arduino IDE?

SpenceKonde, 2020-09-18:
converting it to a an executable would be amazing, would be much cleaner... 

but we don't need to go to that level of work - we can just pull in python like the esp8266 core does!

I know how to do everything except including all the libraries needed by pyupdi.... Am not a python person...

mikkel75, 2020-09-18:
Sounds interresting! Could you elaborate?

I'm running on my Pi directly - ie. in a Raspbian/Linux environment. Installing the libs is easy (see pyupdi main page)

"pip3 install https://github.com/mraardvark/pyupdi/archive/master.zip"

Then you can do sometinh like:

"python3 ~/.local/lib/python3.7/site-packages/updi/pyupdi.py"

Works for me - but would like to have the option inside the Arduino IDE....

mraardvark, 2020-09-18:
FYI: the pyupdi "backend" is now incorporated in pymcuprog, which is installable from pypi:
`pip install pymcuprog`
The command syntax is a bit more generic since this tool is used for many purposes, but mainly native HID debugger interaction.
example: pymcuprog write -d atmega4809 -f hexfile.hex -t uart -u COM1
Right now there are only a few AVRs "supported" (those on CNANOs or IOT boards) but more will be available.
Maintaining pymcuprog (although arguably as a centralised dictatorship :/ ) is easier in my opinion...
Making binaries out of python packages is non-trivial, but not that hard either - although I think performance might be questionable since (eg: pyinstaller) binaries are expanded with all dependencies into a temp folder and executed from there.

qbolsee, 2020-10-28:
About producing a binary from `pyupdi.py`, I quickly tried `pyinstaller --onefile pyupdi.py` on Windows and it produced a single .exe file of around 6Mb. I'm sure this can be reduced a lot, as there are very little non-native dependencies here (only `pyserial`, I believe). I successfully flashed an ATtiny1614 with this .exe file, so I think it's a valid approach. This seems to be what Espressif did with their `esptool.py` tool for the ESP32 Arduino IDE integration.

Maintaining a multi-platform binary release of pyupdy is not necessarily a lot of work: pyinstaller can be integrated with github to produce binaries on its own. Here's an example of a pure-python app with automated binary releases: [https://github.com/esphome/esphome-flasher](https://github.com/esphome/esphome-flasher).

On the other hand, we only really need _one_ working version that @SpenceKonde could integrate in e.g. megaTinyCore. I can give you a hand on the Python side if you think the Arduino IDE integration is straightforward after this. We surely need to test it on Linux platforms as well (how does the Arduino board manager download the right tools for the right OS? I'm not familiar with that part).

SpenceKonde, 2020-10-28:
Wow, if we can produce a single executable, that's perfect! I'm not particularly concerned about the size of it, but if you want to optimize that, I won't stop you!

Build that for as many of the recognized platforms as is possible, compress as appropriate (.zip also works on windows, and .tar.gz is very common for unix-like OS's, host somewhere (doesn't github let you upload supporting binaries - for this sort of purpose?), the metadata gets put into an entry like this that goes in the tools section of the board manager json:

```
        {
          "name": "avrdude",
          "version": "6.3.0-arduino17",
          "systems": [
            {
              "size": "219631",
              "checksum": "SHA-256:2a8e68c5d803aa6f902ef219f177ec3a4c28275d85cbe272962ad2cd374f50d1",
              "host": "arm-linux-gnueabihf",
              "archiveFileName": "avrdude-6.3.0-arduino17-armhf-pc-linux-gnu.tar.bz2",
              "url": "http://downloads.arduino.cc/tools/avrdude-6.3.0-arduino17-armhf-pc-linux-gnu.tar.bz2"
            },
            {
              "size": "229852",
              "checksum": "SHA-256:6cf948f751acfe7b96684537f2291c766ec8b54b4f7dc95539864821456fa9fc",
              "host": "aarch64-linux-gnu",
              "archiveFileName": "avrdude-6.3.0-arduino17-aarch64-pc-linux-gnu.tar.bz2",
              "url": "http://downloads.arduino.cc/tools/avrdude-6.3.0-arduino17-aarch64-pc-linux-gnu.tar.bz2"
            },
            {
              "size": "279045",
              "checksum": "SHA-256:120cc9edaae699e7e9ac50b1b8eb0e7d51fdfa555bac54233c2511e6ee5418c9",
              "host": "x86_64-apple-darwin12",
              "archiveFileName": "avrdude-6.3.0-arduino17-x86_64-apple-darwin12.tar.bz2",
              "url": "http://downloads.arduino.cc/tools/avrdude-6.3.0-arduino17-x86_64-apple-darwin12.tar.bz2"
            },
            {
              "size": "254271",
              "checksum": "SHA-256:accdfb920af2aabf4f7461d2ac73c0751760f525216dc4e7657427a78c60d13d",
              "host": "x86_64-linux-gnu",
              "archiveFileName": "avrdude-6.3.0-arduino17-x86_64-pc-linux-gnu.tar.bz2",
              "url": "http://downloads.arduino.cc/tools/avrdude-6.3.0-arduino17-x86_64-pc-linux-gnu.tar.bz2"
            },
            {
              "size": "244550",
              "checksum": "SHA-256:5c8cc6c17db9300e1451fe41cd7178b0442b4490ee6fdbc0aed9811aef96c05f",
              "host": "i686-linux-gnu",
              "archiveFileName": "avrdude-6.3.0-arduino17-i686-pc-linux-gnu.tar.bz2",
              "url": "http://downloads.arduino.cc/tools/avrdude-6.3.0-arduino17-i686-pc-linux-gnu.tar.bz2"
            },
            {
              "size": "328460",
              "checksum": "SHA-256:e99188873c7c5ad8f8f906f068c33600e758b2e36cce3adbd518a21bd266749d",
              "host": "i686-mingw32",
              "archiveFileName": "avrdude-6.3.0-arduino17-i686-w64-mingw32.zip",
              "url": "http://downloads.arduino.cc/tools/avrdude-6.3.0-arduino17-i686-w64-mingw32.zip"
            }
          ]
        },
```

and then versions of the core that support it include it in their toolsDependencies section, with the other tools they pull in....

```
          "toolsDependencies": [
            {
              "packager": "arduino",
              "name": "avr-gcc",
              "version": "7.3.0-atmel3.6.1-arduino7"
            },
            {
              "packager": "arduino",
              "name": "avrdude",
              "version": "6.3.0-arduino17"
            },
            {
              "packager": "arduino",
              "name": "arduinoOTA",
              "version": "1.3.0"
            }
          ]
```

qbolsee, 2020-10-29:
Alright, I just built an alpha release on my own fork:
https://github.com/qbolsee/pyupdi-bin/releases/tag/v0.1

Let's carefully test this locally before planning a release to the board manager, right now I only tested the windows one. Do you already know how to pass the arguments (COM port, .hex filename) to the binary from the Arduino IDE?

mraardvark, 2020-10-29:
Nice initiative :)
The complications always come with MacOs though, right?
I have no hardware to test on right now, but your binaries in v0.1 work in Linux, but fail on Mac

qbolsee, 2020-10-29:
Yes, it might take a few days to get the MacOS version working as I can't easily test it. It was built with `macOS-latest` in the github workflow, perhaps that can have an effect?

Anyway thanks for the feedback. I think the file format was wrong, I updated it and now it's just a .tar.bz2 with a binary inside:
https://github.com/qbolsee/pyupdi-bin/releases/download/v0.1/pyupdi-bin-v0.1-macOS-x64.tar.bz2

MCUdude, 2020-12-17:
@qbolsee could you update your binaries with PR #73 included? I'd like to experiment with bundling the Win_x64, linux and MacOS binaries with [MegaCoreX](https://github.com/MCUdude/MegaCoreX), which is an Arduino hardware package for the megaAVR-0 chips.

qbolsee, 2020-12-21:
Oh that sounds great, I'd be glad to help! Actually @SpenceKonde and I found a "cleaner" approach for his [megaTinyCore](https://github.com/SpenceKonde/megaTinyCore): we include a portable Python binary and the Python scripts. By adapting `platform.txt`, we can get the portable Python to launch our script. Instead of using pyupdi, we migrated to the more complete [pymcuprog](https://test.pypi.org/project/pymcuprog/), and I wrote an interfacing script to make the Arduino IDE integration easier.

If you think the same approach would be OK for you, we can surely help. If you prefer using pyinstaller-style binaries I can also help, but I have to warn you: Windows Defender is constantly flagging the binary as a trojan. It seems to be a common issue with pyinstaller: [https://stackoverflow.com/questions/54733909/windows-defender-alert-users-from-my-pyinstaller-exe](https://stackoverflow.com/questions/54733909/windows-defender-alert-users-from-my-pyinstaller-exe). I will update my binaries anyway, in case someone finds them useful.

mraardvark, 2020-12-21:
FYI: new pymcuprog released on pypi - although release testing exposed a few 'pyupdi' bugs which we will need to address soon.
Also FYI: I think pyinstaller binaries need signing on Windows and Mac "for best results".