mraardvark/pyupdi #43: hardware vs software serial

neilger, 2019-12-29:
I'm delighted to find pyupdi; I'm looking at options for making embedded interfaces. I've got it working through an FTDI FT230X, but not through the preferred solution of an ATSAMD11C running a USB-UART bridge. I'd like to review what pyupdi needs to be able to configure in the serial interface that's being missed by the simple byte bridge code I tested.

ClundXIII, 2020-01-08:
@neilger I am sorry to hijack your Thread but what was your exact circuit? Does 4.7kOhms work as well or did you go higher or lower? Did you use the 3.3V as VCCIO?

neilger, 2020-01-09:
Hijack away.

This D11 board: 

http://academy.cba.mit.edu/classes/embedded_programming/D11C/hello.D11C.serial.3V3.png

works to do a serial echo:

http://academy.cba.mit.edu/classes/embedded_programming/D11C/hello.D11C.serial.3V3.mp4

to a t1614:

http://academy.cba.mit.edu/classes/embedded_programming/t1614/hello.t1614.echo.png

This FT230XS board:

http://academy.cba.mit.edu/classes/embedded_programming/FTDI/USB-FT230XS-UPDI.png

works to program the t1614:

http://academy.cba.mit.edu/classes/embedded_programming/t1614/hello.t1614.echo.mp4

as does this adapter with an FTDI cable:

http://academy.cba.mit.edu/classes/embedded_programming/UPDI/FTDI-UPDI.png

but not with the D11C bridge.

Looking through the pyupdi code I see it's dynamically reconfiguring the port; short of working through the UPDI protocol, my question is what the bridge code needs to support.

mraardvark, 2020-01-13:
UPDI uses UART with even parity and 2 stop bits.  Baud rate is not that important since the target will sync with it.  In order to recover from an error state, a sequence of two break characters is sent.  This is currently done by reconfiguring the UART to a lower baud and sending slow 0-characters.

neilger, 2020-01-13:
Thanks, that's what I needed, I'll test.