mraardvark/pyupdi #45: Writing User Row

Karibart, 2020-02-05:
Hi
Is there a way to write the User Row of the EEPROM with pyupdi?
Or is there a reason why adding this would be a bad idea? I could give it a try.
Thanks!

mraardvark, 2020-04-03:
Sorry, didn't see this until now.  You can always add it as an optional action.

MarkR42, 2020-07-18:
Yes there is a way to write the user row, write to fuses >= 128, the user row is 0x80 bytes after the fuses 

So pyupdi.py -fs 128:0x2a sets USERROW0 to 0x2a

popovec, 2020-08-11:
The USERROW address in DA devices:  0x1080, FUSES:  0x1050 so the function for fuse writing can be used to write into USERROW, but  LOCK at address 0x1040 can not be accessed in this way. Or did I miss something?