mraardvark/pyupdi #50: AttributeError: 'UpdiApplication' object has no attribute 'progmode'

metropt, 2020-04-24:
Hi,
I can Flash and read the FUSEs but when I try to set a fuse it always fail with the error below.

```
ze-mac:device ze$ pyupdi.py -fs 0:0x00 -c /dev/cu.SLAB_USBtoUART -d tiny202 -b 115200
Device info: {'family': 'tinyAVR', 'nvm': 'P:0', 'ocd': 'D:0', 'osc': '3', 'device_id': '1E9123', 'device_rev': '0.1'}
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.7/bin/pyupdi.py", line 182, in <module>
    _main()
  File "/Library/Frameworks/Python.framework/Versions/3.7/bin/pyupdi.py", line 109, in _main
    if not _process(nvm, args):
  File "/Library/Frameworks/Python.framework/Versions/3.7/bin/pyupdi.py", line 131, in _process
    if not _set_fuse(nvm, fusenum, value):
  File "/Library/Frameworks/Python.framework/Versions/3.7/bin/pyupdi.py", line 163, in _set_fuse
    nvm.write_fuse(fusenum, value)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/updi/nvm.py", line 135, in write_fuse
    return self.application.write_fuse(fuse_address, fuse_data)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/updi/application.py", line 406, in write_fuse_v0
    if not self.progmode:
AttributeError: 'UpdiApplication' object has no attribute 'progmode'
ze-mac:device ze$
```

mraardvark, 2020-04-26:
Sorry, poor test coverage after some feature-creep :/
Doing some refactoring and will hopefully fix this...

metropt, 2020-04-26:
Feel to ask me to test it if you need

nieldk, 2020-06-15:
Any update on this ? I tried to figure out the cause, but sadly unsuccessfull

mraardvark, 2020-06-15:
Nope, sorry - still in my inbox :/ One of those things I should really have done during lockdown...
You are probably best checking out a branch from before that last merge for working fuse support in the meantime...

nieldk, 2020-06-15:
I did that already, and applied a few patches (fuse reading and more devices). Its working. OK, I will try and help with this also. Great program! and basically the only one that reliably works with Leonerds 12V programmer.