mraardvark/pyupdi #54: Refactor AVR DA support

mraardvark, 2020-05-01:
right now its a mess, and should use subclasses for the 16 and 24 bit UPDI modes as well as the two NVM controllers

SpenceKonde, 2020-10-28:
By the way, if you weren't aware, you have to always use 24-bit commands on these parts. Once you do a 24-bit ST ptr, if you do then do a 16-bit STS/LDS, the high byte of the pointer gets used! We ran into that on jtag2updi, It would write a page of flash, then try to write a NOOP to NVMCTRL.CTRLA with a 16-bit STS..... instead writing to a byte of flash at 0x1000 in the flash address space.... and then it would check NVMCTRL.STATUS and see the 0xFF of empty flash.... I was going crazy, heh...

mraardvark, 2020-10-28:
yes, this is by design, and is also the case with xmega PDI...

SpenceKonde, 2020-11-01:
It would have been really nice if the datasheet said as much... I spent weeks trying to figure that out....