mraardvark/pyupdi #56: INFO:link UPDI not OK at programming ATtiny1604

margemag, 2020-05-26:
Hi, 
I'm trying to write a hex on ATtiny1604 using pyupdi.py but getting "INFO:link UPDI not OK"
The host OS is Windows10
The serial adapter is USB -> TTL. 

$ pyupdi.py -d tiny1604 -c COM4 -b 9600 -f Blink.ino.hex -v
INFO:phy Opening COM4 at 9600 baud
INFO:phy send 1 bytes
INFO:phy data:  : [0x0]
INFO:link STCS 0x08 to 0x03
INFO:phy send 3 bytes
INFO:phy data:  : [0x55, 0xc3, 0x8]
INFO:link STCS 0x80 to 0x02
INFO:phy send 3 bytes
INFO:phy data:  : [0x55, 0xc2, 0x80]
INFO:link LDCS from 0x00
INFO:phy send 2 bytes
INFO:phy data:  : [0x55, 0x80]
INFO:phy receive : []
INFO:link UPDI not OK - reinitialisation required
INFO:phy Sending double break
INFO:phy Opening COM4 at 9600 baud
INFO:link STCS 0x08 to 0x03
INFO:phy send 3 bytes
INFO:phy data:  : [0x55, 0xc3, 0x8]
INFO:link STCS 0x80 to 0x02
INFO:phy send 3 bytes
INFO:phy data:  : [0x55, 0xc2, 0x80]
INFO:link LDCS from 0x00
INFO:phy send 2 bytes
INFO:phy data:  : [0x55, 0x80]
INFO:phy receive : []
INFO:link UPDI not OK - reinitialisation required
Traceback (most recent call last):
  File "C:\python\pyupdi-master\pyupdi.py", line 182, in <module>
    _main()
  File "C:\python\pyupdi-master\pyupdi.py", line 95, in _main
    nvm = UpdiNvmProgrammer(comport=args.comport,
  File "C:\python\pyupdi-master\updi\nvm.py", line 17, in __init__
    self.application = UpdiApplication(comport, baud, device)
  File "C:\python\pyupdi-master\updi\application.py", line 17, in __init__
    self.datalink = UpdiDatalink(comport, baud)
  File "C:\python\pyupdi-master\updi\link.py", line 32, in __init__
    raise Exception("UPDI initialisation failed")
Exception: UPDI initialisation failed
INFO:phy Closing port 'COM4'

What I am doing incorrect? Seems it supposed to work for tiny1604

r00sta, 2020-05-26:
Have you set it into programming mode with a 12V pulse?

margemag, 2020-05-26:
No, I haven't set it into UPDI mode with 12V pulse but I think it already in UPDI mode as I can flesh in it using Arduino / megaTinyCore.

r00sta, 2020-05-26:
What is your updi setup?

margemag, 2020-05-26:
I am not sure about flashing the Arduino bootloader, but until now I am using Arduino without issues to flash on tiny1604.
As tiny1604's doc says "UPDI is based on standard UART communication, using a fixed frame
format, and automatic baud rate detection" I think it had to work with programmer's set baud rate, or I am mistaken.
Can you tell please how can I get details about UPDI setup?

r00sta, 2020-05-27:
By UPDI setup I meant that have you connected the serial adapter as shown in the README of this repo? What serial adapter are you using? Have you tried different baud rates?

An alternative could be to use an arduino as the uart serial bridge https://github.com/SpenceKonde/megaTinyCore/blob/master/MakeUPDIProgrammer.md

Finally I would recommend this programmer https://www.tindie.com/products/leonerd/avr-updi-programmer-with-12v/ (it's unfortunately out of stock but all the files are open source)