mraardvark/pyupdi #60: " ihex too large for flash"

shyhkaelo, 2020-07-06:
I built a project and compiled it, the program used 520 bytes showed in MPLAB X IDE
But when I executed pyupdi to download my code, terminal replied the information " **ihex too large for flash** "
[image]

edenraf, 2021-04-22:
A year later...did anyone find a reason for this or a fix? I have the same issue. MPLAB IDE tells me I'm using a few % and then pyupdi says its too big

mraardvark, 2021-04-22:
Same thought - MPLAB is putting fuses in the hexfile at a huge offset, and pyupdi is not fuse-aware.
I recommend to use pymcuprog which is properly maintained
Give it a try at least: pip install pymcuprog

edenraf, 2021-04-22:
> Same thought - MPLAB is putting fuses in the hexfile at a huge offset, and pyupdi is not fuse-aware.
> I recommend to use pymcuprog which is properly maintained
> Give it a try at least: pip install pymcuprog

Hey thanks for the reply. I'll check it out. I am trying to simply program with it but I have no luck. Is this correct?
`pymcuprog write -d mega808 -t UART -u COM10 -f myfile.hex`

I get the answer:
`Connecting to any UART
pyedbglib.hidtransport.hidtransportbase - ERROR - No CMSIS-DAP devices found.
Unable to connect to USB device`

edenraf, 2021-04-22:
> Hey thanks for the reply. I'll check it out. I am trying to simply program with it but I have no luck. Is this correct?
> `pymcuprog write -d mega808 -t UART -u COM10 -f myfile.hex`
>
> I get the answer: [...]

For anyone else facing the problem of not reading the documentation well, I did this and it worked
`pymcuprog ping -d atmega808 -t uart -u com10`

mraardvark, 2021-04-22:
ah, case sensitive...