mraardvark/pyupdi #62: UPDI initialisation failed - ATTINY402

saurabhjhamb, 2020-07-29:
Hi Folks,

Can anyone help me with following error screen shots?

Setup TTL to uart converter: CH340G
TX connected to RX with 4.7K connected to UPDI (PIN6)
VCC of UART & ATT402 shorted to 5V
GND: ATT402 shorted to TTL GND

At power on, connecting a momentary pulse of 12V to UPDI pin and then running following command

command: python pyupdi.py -d tiny402 -f att402.hex -c COM4 -v
result:
INFO:phy Opening COM4 at 115200 baud
INFO:phy send 1 bytes
INFO:phy data:  : [0x0]
INFO:link STCS 0x08 to 0x03
INFO:phy send 3 bytes
INFO:phy data:  : [0x55, 0xc3, 0x8]
INFO:link STCS 0x80 to 0x02
INFO:phy send 3 bytes
INFO:phy data:  : [0x55, 0xc2, 0x80]
INFO:link LDCS from 0x00
INFO:phy send 2 bytes
INFO:phy data:  : [0x55, 0x80]
INFO:phy receive : []
INFO:link UPDI not OK - reinitialisation required
INFO:phy Sending double break
INFO:phy Opening COM4 at 115200 baud
INFO:link STCS 0x08 to 0x03
INFO:phy send 3 bytes
INFO:phy data:  : [0x55, 0xc3, 0x8]
INFO:link STCS 0x80 to 0x02
INFO:phy send 3 bytes
INFO:phy data:  : [0x55, 0xc2, 0x80]
INFO:link LDCS from 0x00
INFO:phy send 2 bytes
INFO:phy data:  : [0x55, 0x80]
INFO:phy receive : []
INFO:link UPDI not OK - reinitialisation required
Traceback (most recent call last):
  File "pyupdi.py", line 182, in <module>
    _main()
  File "pyupdi.py", line 95, in _main
    nvm = UpdiNvmProgrammer(comport=args.comport,
  File "d:\Sanitizer\pyupdi\updi\nvm.py", line 17, in __init__
    self.application = UpdiApplication(comport, baud, device)
  File "d:\Sanitizer\pyupdi\updi\application.py", line 17, in __init__
    self.datalink = UpdiDatalink(comport, baud)
  File "d:\Sanitizer\pyupdi\updi\link.py", line 32, in __init__
    raise Exception("UPDI initialisation failed")
**Exception: UPDI initialisation failed**
INFO:phy Closing port 'COM4'

maciejap, 2020-08-05:
Hi! 
"At power on, connecting a momentary pulse of 12V to UPDI pin and then running following command"
If You put 12V on RX of CH340 then can be damaged. If is not broken (your converter), mayby clamping diode (of  ch340) pull down 12V to USB, then 12V not present on UPDI.
When i use converter based on CH340 and powered attiny by 5V i sometimes have problems with initialization and failure in midle of programming. If i use 3.3V everythink was work fine.

saurabhjhamb, 2020-08-05:
@maciejap You are right! that's why I do NOT connect RX of CH340 to 12V.
I manually provide 12V pulse on UPDI (without RX connected) and then connect RX of CH340 (after removing 12v).

Same error continues to show up...

maciejap, 2020-08-05:
In the chapter 30.3.2.2 "UPDI Enable with 12V Override of RESET Pin" in the datasheet for attiny402 is described timings for eneable UPDI by 12V.
Mayby You shoudl look for request "HV Mode" on this github.

saurabhjhamb, 2020-08-06:
Yes, thanks for pointing it out.
The timing suggests to give a pulse of 100 μs to 1 ms, that might not be possible with manual control.
Seems like I will have to use a pulse / function generator.

Is there any easier way to be able to use UPDI?
How else PYUPDI is expected to help?

maciejap, 2020-08-06:
For now You can use pyUPDI when pin UPDI is not configured as gpio (when is fused as !RESET, 12V is not required).
If You must use HV mode that pulled request is for You:
https://github.com/mraardvark/pyupdi/pull/37.

saurabhjhamb, 2020-08-06:
OK, but that's an issue still.

I run pyUPDI after a power on (RX of CH340 is connected to RESET pin) I get the same error logs.
It doesn't work.

maciejap, 2020-08-13:
You are sure that attiny and ch340 is not broken? Do you set on attiny any fuse/lock bits?
Can you shown your circuit diagram with attiny and connected programmer.