mraardvark/pyupdi #66: AtTiny404 link UPDI not OK Reinitialization required

dreadlordrider, 2020-09-29:
I know this issue has been raised for different devices, but I feel in my case I have a different condition.
I am using a Raspberry Pi and a FT232R USB UART adapter. The Tiny does not have the UPDI pin configured as an i/o pin. Yet it keeps giving me the error that it failed to be initialized.
I have tried both the GPIO serial and USB2TTL. I changed the GPIO setup according to the link too. but no success.
My terminal output is 
` python3 pyupdi.py -v -d tiny404 -c /dev/ttyUSB0 -b 115200 -f /home/pi/Desktop/test.hex
INFO:phy Opening /dev/ttyUSB0 at 115200 baud
INFO:phy send : [0x0]
INFO:link STCS to 0x03
INFO:phy send : [0x55, 0xc3, 0x8]
INFO:link STCS to 0x02
INFO:phy send : [0x55, 0xc2, 0x80]
INFO:link LDCS from 0x00
INFO:phy send : [0x55, 0x80]
INFO:phy receive : []
INFO:link UPDI not OK - reinitialisation required
INFO:phy Sending double break
INFO:phy Opening /dev/ttyUSB0 at 115200 baud
INFO:link STCS to 0x03
INFO:phy send : [0x55, 0xc3, 0x8]
INFO:link STCS to 0x02
INFO:phy send : [0x55, 0xc2, 0x80]
INFO:link LDCS from 0x00
INFO:phy send : [0x55, 0x80]
INFO:phy receive : []
INFO:link UPDI not OK - reinitialisation required
Traceback (most recent call last):
  File "pyupdi.py", line 178, in <module>
    _main()
  File "pyupdi.py", line 95, in _main
    device=Device(args.device))
  File "/home/pi/Downloads/pyupdi-master/updi/nvm.py", line 17, in __init__
    self.application = UpdiApplication(comport, baud, device)
  File "/home/pi/Downloads/pyupdi-master/updi/application.py", line 17, in __init__
    self.datalink = UpdiDatalink(comport, baud)
  File "/home/pi/Downloads/pyupdi-master/updi/link.py", line 31, in __init__
    raise Exception("UPDI initialisation failed")
Exception: UPDI initialisation failed
INFO:phy Closing /dev/ttyUSB0`

Any help or suggestions will be greatly appreciated.

dreadlordrider, 2020-09-29:
python3 pyupdi.py -v -d tiny404 -c /dev/ttyUSB0 -b 115200 -f /home/pi/Desktop/test.hex
INFO:phy Opening /dev/ttyUSB0 at 115200 baud
INFO:phy send : [0x0]
INFO:link STCS to 0x03
INFO:phy send : [0x55, 0xc3, 0x8]
INFO:link STCS to 0x02
INFO:phy send : [0x55, 0xc2, 0x80]
INFO:link LDCS from 0x00
INFO:phy send : [0x55, 0x80]
INFO:phy receive : []
INFO:link UPDI not OK - reinitialisation required
INFO:phy Sending double break
INFO:phy Opening /dev/ttyUSB0 at 115200 baud
INFO:link STCS to 0x03
INFO:phy send : [0x55, 0xc3, 0x8]
INFO:link STCS to 0x02
INFO:phy send : [0x55, 0xc2, 0x80]
INFO:link LDCS from 0x00
INFO:phy send : [0x55, 0x80]
INFO:phy receive : []
INFO:link UPDI not OK - reinitialisation required
Traceback (most recent call last):
  File "pyupdi.py", line 178, in <module>
    _main()
  File "pyupdi.py", line 95, in _main
    device=Device(args.device))
  File "/home/pi/Downloads/pyupdi-master/updi/nvm.py", line 17, in __init__
    self.application = UpdiApplication(comport, baud, device)
  File "/home/pi/Downloads/pyupdi-master/updi/application.py", line 17, in __init__
    self.datalink = UpdiDatalink(comport, baud)
  File "/home/pi/Downloads/pyupdi-master/updi/link.py", line 31, in __init__
    raise Exception("UPDI initialisation failed")
Exception: UPDI initialisation failed
INFO:phy Closing /dev/ttyUSB0

mraardvark, 2020-09-29:
any chance you have access to a scope to see what's happening on the tx/rx lines? :|

dreadlordrider, 2020-09-29:
[image]
It is definitely sending something. I have a video of the attempts but I'm unable to upload it here.

dreadlordrider, 2020-09-29:
[image]

mraardvark, 2020-09-29:
Hmmm - maybe you should install a pullup?

dreadlordrider, 2020-09-29:
So instead of a 4.7k resistor between Tx and UPDI, should I connect Tx, Rx UPDI together and a 4.7k to 3.3V?

dreadlordrider, 2020-09-29:
This is across TxRx
[image]

mraardvark, 2020-09-29:
Looks like the double-break, which is used to "recover" once an error is suspected - any sign of the LDCS/STCS?

mraardvark, 2020-09-29:
Yes, this is a typical try-doublebreak-retry dialog.
Are you scoping TX with respect to GND or UPDI with respect to GND?

dreadlordrider, 2020-09-29:
Here is the scope between TX and GND

[image]

mraardvark, 2020-09-29:
FYI - you might want to try and adjust the trigger to catch one of those shorter negative-pulses as they contain more edges than you can see there, but:
- it looks to me like the device is not responding, or can't be heard
- try scoping UPDI-GND to see if you see the same or an attempted response?
- are you sure you are hooked up to the UPDI pin and it is enabled (not set to GPIO/RESET)?

dreadlordrider, 2020-09-29:
I'm 100% sure it is not set to reset/gpio.
I have used it so far with arduino IDE connected a nano flashed with jtag2updi. Right now I uploaded a basic blink program connected to pin 4 which is PA6. 
With Arduino IDE I have no problem uploading code.
I just moved to RPi and am unable to upload the same exact HEX file that I uploaded before.
I will try the UPDI to GND again. Right now I had taken all my connections out and was trying different circuits.

mraardvark, 2020-09-29:
OK.  Last question before I sign off:
- is that y axis correct on the display?  100mV per division makes this signal less than 400mV??  
Or am I reading it wrong.

dreadlordrider, 2020-09-29:
You are definitely right. it is 320mV 
I just assumed that was how low the voltages are supposed to be. I have it plugged directly into the Pi. The Pi has a 5V2A power source

dreadlordrider, 2020-09-29:
It maybe an issue with the probe first thing tomorrow I'll change the probe and test it again.

mraardvark, 2020-09-30:
No progress? Been able to catch any frames on the UPDI side (not TX pin)?

dreadlordrider, 2020-09-30:
Nothing from UPDI. Right now I'm at the point of giving up on the PyUPDI and going for the Pi->Nano->Tiny via jtag2updi

mraardvark, 2020-10-01:
Best do what works for you.
I guess you have tried the PI UART directly on its pins?

dreadlordrider, 2020-10-01:
Yup, I tried both methods. Serial GPIO and usb2serial.
😭

mraardvark, 2020-10-01:
can you capture and post the UPDI init sequence - not the double break - it should consist of about 3 short frames starting with a SYNC on each.

dreadlordrider, 2020-10-01:
So I get 3 negative peaks. then the double break. then sometimes 2 sometimes 3 peaks. Maybe its always 3, I just cant see it some times
I tried changing the trigger, yes I can see multiple 1s and 0s. but cant capture it.