mraardvark/pyupdi #72: Support the same target names as Avrdude

MCUdude, 2020-12-17:
For integration with other tools that uses avr-gcc and Avrdude (Like Arduino IDE and PlatformIO) would be great if pyupdi supported the same target names as avr-gcc and Avrdude uses.

In other words, it would be great if we also could refer to chips as `atmega4809` and not _just_ `mega4809`.

As you can see in the PlatformIO configuration for pyupdi uploads below, `mega4809` actually has to be "manually" specified in the upload flags. It would be much more elegant if it could use the underlying `atmega4809` name instead so I could refer to the chip with `${board_build.mcu}` instead.

```ini
[env:pyupdi_upload]
platform = atmelmegaavr
framework = arduino
board = ATmega4809
upload_speed = 115200
upload_flags =
  -d
  mega4809
  -c
  $UPLOAD_PORT
  -b
  $UPLOAD_SPEED
upload_command = pyupdi $UPLOAD_FLAGS -f $SOURCE
```

Would you accept a PR where "extra" attiny and atmega targets are added in device.py?

janegilruud, 2020-12-17:
...or make the 'at'-prefix optional on tiny and mega parts?

MCUdude, 2020-12-17:
> ...or make the 'at'-prefix optional on tiny and mega parts?

Yes, even better!

mraardvark, 2020-12-17:
be my guest 👍

MCUdude, 2020-12-17:
> ...or make the 'at'-prefix optional on tiny and mega parts?

> be my guest 👍

As a guy who has barely written python code, this may be a fun challenge.

I'll guess that `device_name` in device.py is a string, and can be checked if it starts with `at*`? If so, it looks like it's pretty straight forward to shave off these two characters and continue on with the process.

MCUdude, 2020-12-17:
As I see it, there are two ways #73 could be solved. The first option is to make at* optional as I've done. The other option is to add "duplicates" to the DEVICES lists. What do you prefer @mraardvark?

mraardvark, 2020-12-17:
optional works for me.  
actually, i would also be happy to hard-rename the old ones to "correct" names, but i doubt users will like that...
(in the end, there will* be no more tiny/mega names coming out as all new parts start with "AVR", so at least this problem is not growing)

MCUdude, 2020-12-17:
> actually, i would also be happy to hard-rename the old ones to "correct" names, but i doubt users will like that...

We could #73 the other way around. Rename all targets to atmega* and attiny*, but still support mega* and tiny*. You're in charge!

mraardvark, 2020-12-17:
yes, rename and pre-pend the AT.... but be careful not to prepend AT in front of the AVRxxD etc!

janegilruud, 2020-12-17:
Only prepend if name starts with 'tiny' or 'mega'.

MCUdude, 2020-12-17:
```python
# Add add at* prefix if not present
if(device_name.startswith("tiny") or device_name.startswith("mega")):
    device_name = "at" + device_name
```

MCUdude, 2020-12-17:
#71 updated!

mraardvark, 2020-12-17:
<insert slight delay due to Julelunsj>

MCUdude, 2020-12-17:
Enjoy your meal! We're having julelønsj on Tuesday 😉

MCUdude, 2020-12-17:
#73 merged. Thank you for your fast response!

mraardvark, 2020-12-17:
No problem - anything is possible with some ribbe... :)
Actually this was requested by other users on another forum and was long overdue

closed 2020-12-17 by MCUdude