mraardvark/pyupdi #74: Can't write fuses to AVR128DB32

Miraculix200, 2021-01-18:
With pyupdi I can successfully write a bootloader to the device (at least it says that it was successful) and read the fuses. However it does not successfully set the fuses.

When doing something like 
` python.exe pyupdi.py -d avr128db32 -c COM3 -fs 2:0x01 -b 9600 -v`

I get this response:
```
Verify error for fuse 2, expected 0x01 read 0x00
Error during processing
```

The fuses simply stay the same as they were before.

Any idea what the problem could be? When setting fuses of a ATmega4808 I don't have any problems.

mraardvark, 2021-01-18:
Hi,
Could you perhaps try pymcuprog as a replacement to pyupdi?
pip install pymcuprog
pymcuprog -t uart -u COM3 -c 9600 -d avr128db32 write -m fuses -l <fuse values>

Miraculix200, 2021-01-18:
I don't know what I just did, but it looks like it wrote something to the fuses memory:
`pymcuprog -t uart -u COM3 -c 9600 -d avr128db32 write -m fuses -l 1
Writing literal values...
Done.`

Miraculix200, 2021-01-18:
pyupdi doesn't like what I did, but the 0 fuse has changed from 0x00 to 0x01

mraardvark, 2021-01-18:
Yes, we could always do with more examples...  once it gets to github it will be easier to add examples there.
Are you able to write flash and other memories you need too then?

Miraculix200, 2021-01-18:
I can now write sketches to it using DxCore on Arduino IDE, after setting the correct fuses (BOOTSIZE and RSTPINCFG). I don't need anything else right now, but I see no reason why it wouldn't work with pymcuprog.

Thanks for the hint

Miraculix200, 2021-01-18:
I'll write a small step by step guide on the page of a open source dev board for AVR*DB32 I created, so maybe you can also give a hint how I can flash a bootloader without destroying the MCU :) 

Otherwise the step by step guide would include both pyupdi and pymcuprog. Though I guess I could also figure it out myself tomorrow.

mraardvark, 2021-01-18:
Sure.
FYI - pymcuprog will be maintained with new devices going forward well before pyupdi; but will always maintain a consistent API (since its used for various purposes)

One oddity (/issue) right now is that the device must be erased in a separate command: pymcuprog erase
After that, you can write a hexfile to flash simply using pymcuprog write -f <hexfile>
Besides that, the "ping" command is a useful "connectivity check" which i run before anything else

Miraculix200, 2021-01-19:
That also worked. So I guess unlike pyupdi, pymcuprog can do everything it's supposed to.

closed 2021-02-19 by Miraculix200