mraardvark/pyupdi #76: How to use pyupdi

paulskirk53, 2021-02-13:
Hello, I read the readme file and I am looking to program atmega 4809 which uses updi as a programming method. Also using Platformio IDE. When I look at the pyupdi circuit diagram, it looks like an FTDI board on the left, but my question is, could an arduino be used instead of the FTDI? If it's possible to use an Arduino Uno, what software would be needed on the Uno?

mraardvark, 2021-02-13:
Hi Paul - I guess your board has some sort of USB-serial converter, which is technically all you need.  However you will need to prevent the mega328(?) device from interfering  with UPDI, so I would guess it has to be erased/disconnected somehow.  I am not an expert with the Arduino boards unfortunately...

paulskirk53, 2021-02-13:
thanks for reply. It's probably just as easy to use a USB to serial converter, I'll try that. Do you by any chance know if it's possible to debug the 4809 chip using UPDI? I'm new to MCUs outside of Arduino and using Platformio for coding. Thanks very much,
Paul

mraardvark, 2021-02-13:
You can debug using UPDI, but not with pyupdi - the debug system is not exposed.
I would recommend getting a Snap as standalone debugger or Curiosity Nano kit with on-board debugger (can easily be cut off and used as a standalone debugger, but for UPDI parts only).  The platformio support for debug is (as I have tried) still clumsy compared to Studio or MPLABX.

paulskirk53, 2021-02-13:
thanks very much for the advice. I guess the snap debugger will work with atmel studio 7 as its the same manufacturer, and I'll enquire with the platformio guys if it will work with PIO.
much appreciate your help - it's a minefield for newcomers!
Paul

mraardvark, 2021-02-13:
Snap works with Studio 7 in "AVR mode" (in which it looks and behaves like an atmel-ice) but still has to be upgraded in MPLABX.
Moving to "bare metal" has its challenges, yes... but can be fun and rewarding :)
If you can get your hands on a Curiosity Nano kit it can be a great starting point (although I may be biased here :/ )
Both bare metal and MCC/Start (library-based) examples are now published on github for the new AVRs: https://github.com/microchip-pic-avr-examples so filter by device and look for something interesting there to play with.
I think you will need "debugging", and you will need a decent "peripheral view" which both Studio 7 and MPLAB have (not sure about platformio).  Try it in the simulator first even, without spending a penny...