mraardvark/pyupdi #77: help with circuit

paulskirk53, 2021-02-15:
Hi, I was wondering if you might be able to point me at some help to understand the pyupdi electronic circuit - connecting Rx and Tx with a resistor. What does that do and how does it help with UPDI? The reason for my question is that I'm using an FTDI board for the updi upload. but also want to use the same board for serial comms on a 4809 UART. I posted on the avrfreaks forum and got a circuit diagram in response which I'm working to understand the logic of, but to do that I have to get my head around what happens when the Tx and Rx on a serial port are connected with a 4k7 resistor. Many thanks for your help,
Paul
btw pyupdi works fine to upload code to my 4809 - thanks.

mraardvark, 2021-02-15:
Hi Paul - its no magic, its just contention-mitigation.  UPDI is bidirectional on a single pin while the CDC uses two pins.  Communications start with the CDC sending over TX (driving high and low) and UPDI will respond.  The CDC's idle state is '1', so the TX line can  technically still be driving this '1' state out while the UPDI response comes back.  Putting the resistor in place will allow the UPDI response through to RX despite TX driving '1'.  The resistor value thus depends somewhat on drive-strength of your CDC-USB adapter...

paulskirk53, 2021-02-15:
Thanks very much for this - much appreciated, my electronics knowledge is just about up to understanding what you've said :). So if I've got it right, CDC Tx is used to 'activate' UPDI on the target and UPDI Tx and Rx take place on the one wire. The resistor (cleverly) allows it to be a one wire comms system keeping Tx and Rx separate.