mraardvark/pyupdi #79: After updating to the latest version, I can't flash my AVR64DA28

avrnerdz, 2021-03-12:
Traceback (most recent call last):
  File "/usr/local/bin/pyupdi", line 8, in <module>
    sys.exit(_main())
  File "/usr/local/lib/python3.8/dist-packages/updi/pyupdi.py", line 111, in _main
    if not _process(nvm, args):
  File "/usr/local/lib/python3.8/dist-packages/updi/pyupdi.py", line 136, in _process
    return _flash_file(nvm, args.flash)
  File "/usr/local/lib/python3.8/dist-packages/updi/pyupdi.py", line 151, in _flash_file
    nvm.write_flash(start_address, data)
  File "/usr/local/lib/python3.8/dist-packages/updi/nvm.py", line 111, in write_flash
    return self._write_mem(address, data, self.device.flash_pagesize, use_word_access=True)
  File "/usr/local/lib/python3.8/dist-packages/updi/nvm.py", line 133, in _write_mem
    self.application.write_nvm(address, page, use_word_access=use_word_access)
TypeError: write_nvm_v1() got an unexpected keyword argument 'use_word_access'
INFO:phy Closing port '/dev/ttyUSB0'
make: *** [makefile:18: upload] Error 1

mraardvark, 2021-03-12:
...a symptom of having no test setup and too much avr-diversity.
Any chance you could use pymcuprog instead?

avrnerdz, 2021-03-12:
I've tried pymcuprog, but I've had strange issues with that (appeared no not be cleaning flashing updated code, perhaps I need to issue a separate erase command?) I'll give it another try.

mraardvark, 2021-03-12:
Good :) 
Erase before write is a much discussed feature...