mraardvark/pyupdi #83: Exception: UPDI initialisation failed

Vepla123, 2021-06-11:
C:\Users\User\pyupdi-master\updi>pyupdi.py -d tiny414 -c COM2 -b 9600 -f LED_414.hex
Traceback (most recent call last):
  File "C:\Users\User\pyupdi-master\updi\pyupdi.py", line 207, in <module>
    _main()
  File "C:\Users\User\pyupdi-master\updi\pyupdi.py", line 97, in _main
    nvm = UpdiNvmProgrammer(comport=args.comport,
  File "C:\Users\Dan\AppData\Roaming\Python\Python39\site-packages\updi\nvm.py", line 17, in __init__
    self.application = UpdiApplication(comport, baud, device)
  File "C:\Users\Dan\AppData\Roaming\Python\Python39\site-packages\updi\application.py", line 17, in __init__
    self.datalink = UpdiDatalink(comport, baud)
  File "C:\Users\Dan\AppData\Roaming\Python\Python39\site-packages\updi\link.py", line 32, in __init__
    raise Exception("UPDI initialisation failed")
Exception: UPDI initialisation failed

Any one know why im failing?

mraardvark, 2021-06-12:
Looks like a "no contact" situation... (cabling, UPDI disabled, electrical issues...)
FYI: please use pymcuprog instead of pyupdi

Vepla123, 2021-06-12:
Thank you. 
I used pymcuprog.  Pinged the chip  and hexed with no problem.