mraardvark/pyupdi #84: attiny424 not programmed using pyupdi.

sahil-siwatch, 2021-07-30:
Hello all. 
I have been trying to use attiny424 (avr-2 series). When i upload it using pyupdi, it shows that attiny424 is notwrong argument. I checked datasheet and it clearly shows that UPDI is available. 

[image]

mraardvark, 2021-07-30:
I recommend to use pymcuprog instead of pyupdi:
https://pypi.org/project/pymcuprog/

sahil-siwatch, 2021-07-30:
I've installed it. when i check in device folder, attiny424 is not present
got any tutorials to use ??

mraardvark, 2021-07-30:
Good point - I was not aware it was missing.
I would just copy the attiny414.py in the device scripts folder and edit the device id to match the 424.
Start by checking connectivity: pymcuprog ping -d attiny424 -t uart -u <comxyz>
then you could for example write flash using: pymcuprog write -m flash -f hexfile.hex

sahil-siwatch, 2021-07-30:
Using UPDI, I can burn code if I select any microcontroller which is listed, I can upload the code this way. Maybe i can select attiny414 and write code..might work..will update.