mraardvark/pyupdi #86: Query re upload time

paulskirk53, 2021-11-16:
Hi, Just observing that my PyUPDI on an AVR4809 takes circa 30 seconds to upload an 11Kbyte program. Is this ok? It works, so no complaints, just seems a relatively long time. Thanks.

mraardvark, 2021-11-16:
quick test with pymcuprog and a canned 16k hexfile takes 2.1 seconds (3.1 with verification)

paulskirk53, 2021-11-16:
Thanks very much for that, I'm using an FTDI and the resistor, the upload speed is 115200. Is there anything I can do to improve ?

paulskirk53, 2021-11-16:
Ah just noticed you mentioned pymcuprog, would it be better to use that?

mraardvark, 2021-11-16:
indeed.  its maintained :)
also use -x to get quick timing output

mraardvark, 2021-11-16:
Ah, of course - sorry, in my haste I tested the native USB programming speed using a CNANO.  Using the serialUPDI mechanism will be slower, but you should be able to go to ~500k baud.
I can't test right now because my box of serialUPDI tricks is at home...

paulskirk53, 2021-11-18:
I have had some help from the PlatformIO IDE forum and I managed to get pymcuprog to work, but upload of the 11k bytes code takes a longish 3 minutes. The max -c parameter I can use which still initialises is 250k....

mraardvark, 2021-11-18:
Hmmm, slow indeed - I think this is a function of the python serial send/receive, which needs some optimising, and could also be affected by the HW adapter's USB handling (whether its optimised for latency on single characters, or optimised to fill USB frames and get the best bandwidth).