mraardvark/pyupdi #87: pyupdi suddenly stops working on arch linux

wimalopaan, 2021-12-28:
I use pyupdi for years on arch linux without problems.

I don't know which update here produces this error, but actually it is broken:

```
~/.local/bin/pyupdi -d avr128da32 -c /dev/ttyUSB0  -f multiAdapter_01.hex                1  ST 3   master
Device info: {'family': 'AVR', 'nvm': 'P:2', 'ocd': 'D:1', 'osc': '3', 'device_id': '1E9709', 'device_rev': '1.7'}
Traceback (most recent call last):
File "/home/lmeier/.local/bin/pyupdi", line 33, in <module>
sys.exit(load_entry_point('pyupdi==0.0.0', 'console_scripts', 'pyupdi')())
File "/home/lmeier/.local/lib/python3.10/site-packages/updi/pyupdi.py", line 111, in _main
if not _process(nvm, args):
File "/home/lmeier/.local/lib/python3.10/site-packages/updi/pyupdi.py", line 136, in _process
return _flash_file(nvm, args.flash)
File "/home/lmeier/.local/lib/python3.10/site-packages/updi/pyupdi.py", line 151, in _flash_file
nvm.write_flash(start_address, data)
File "/home/lmeier/.local/lib/python3.10/site-packages/updi/nvm.py", line 111, in write_flash
return self._write_mem(address, data, self.device.flash_pagesize, use_word_access=True)
File "/home/lmeier/.local/lib/python3.10/site-packages/updi/nvm.py", line 133, in _write_mem
self.application.write_nvm(address, page, use_word_access=use_word_access)
TypeError: UpdiApplication.write_nvm_v1() got an unexpected keyword argument 'use_word_access'
```

mraardvark, 2021-12-28:
@wimalopaan - full support for Dx devices was never added properly before moving over to pymcuprog.  Suggest you use that package instead.

wimalopaan, 2021-12-28:
Thank you. Found that project and it works very well!

mraardvark, 2021-12-28:
Good to hear!

wimalopaan, 2021-12-28:
pymcuprog seems not to work correctly for the DA. It programms only the first time the device corrctly, subsequent flashes do not overwrite existing programs. Strange ...

wimalopaan, 2021-12-28:
To verify I would like to get this piece working again, because it was successful all the time as well with the DA devices.

wimalopaan, 2021-12-28:
Maybe a python-3.10 issue?

mraardvark, 2021-12-28:
are you using --erase or erasing between writes?  (pymcuprog does not erase without being told to)

wimalopaan, 2021-12-28:
Ok, thanks.
I switch to the pre-release with the --erase flag.

closed 2021-12-28 by wimalopaan