mraardvark/pyupdi #88: How do I add new controllers?

mikrocoder, 2024-05-09:
Hi,

I have edited the device.py. Added the following lines.

```
# AVR Ex
DEVICE_AVR_EA64_SERIES = set(["avr64ea28", "avr64ea32", "avr64ea48"])
```
...

```
elif device_name in DEVICE_AVR_EA64_SERIES:
            self.fuses_address = 0x1050
            self.userrow_address = 0x1080
            self.lock_address = 0x1040
            self.flash_start = 0x800000
            self.flash_size = 64 * 1024
            self.flash_pagesize = 128 
```

updi still does not recognize the controller.

CMD command:
`C:\pyupdi>pyupdi.py -c COM8 -d avr64ea32 -v`

```
usage: pyupdi.py [-h] -d
                 {avr128da28,avr128da32,avr128da48,avr128da64,mega3208,mega3209,mega4808,mega4809,tiny1604,tiny1606,tiny1607,tiny1614,tiny1616,tiny1617,tiny202,tiny204,tiny212,tiny214,tiny3216,tiny3217,tiny402,tiny404,tiny406,tiny412,tiny414,tiny416,tiny417,tiny804,tiny806,tiny807,tiny814,tiny816,tiny817}
                 -c COMPORT [-e] [-b BAUDRATE] [-f FLASH] [-r] [-i] [-fs [FUSES ...]] [-fr] [-v]
pyupdi.py: error: argument -d/--device: invalid choice: 'avr64ea32' (choose from 'avr128da28', 'avr128da32', 'avr128da48', 'avr128da64', 'mega3208', 'mega3209', 'mega4808', 'mega4809', 'tiny1604', 'tiny1606', 'tiny1607', 'tiny1614', 'tiny1616', 'tiny1617', 'tiny202', 'tiny204', 'tiny212', 'tiny214', 'tiny3216', 'tiny3217', 'tiny402', 'tiny404', 'tiny406', 'tiny412', 'tiny414', 'tiny416', 'tiny417', 'tiny804', 'tiny806', 'tiny807', 'tiny814', 'tiny816', 'tiny817')
```
I can flash an ATmega4809. Basically it works.
Python 3.12 is installed.
How do I add new controllers?

mraardvark, 2024-05-09:
I suggest you just use pymcuprog as per the deprecation notice here.
https://pypi.org/project/pymcuprog/

mikrocoder, 2024-05-09:
Thank you. I had already installed 'pymcuprog' but hadn't remembered. Many thanks for your work.

closed 2024-05-09 by mikrocoder